softwaredownload/openwrt-fanqiang #7: opernwrt的shadowsocks客户端，可以和shadowsocks 服务器 python版兼容吗？

travelbob2005, 2015-08-24:
你好，请问openwrt的shadowsocks可以连接shadowsocks python版服务器吗？

travelbob2005, 2015-08-24:
另外shadowsocks libev版的客户端有windows版的嘛？

wickila, 2015-08-24:
1，openwrt的shadowsocks链接python的shadowsocks服务器完全没问题。
2，不太明白你要shadowsocks libev的windows版本做什么，因为windows上有很好用的shadowsocks的GUI客户端。

travelbob2005, 2015-08-24:
明白了，第二个问题是多余了，但是我用operwrt里的shadowsocks连接shadowsocks服务器python版的不行，但是连接shadowsocks libev版的服务器，没问题，shadwoscks服务器pyhon版的，用shadowsock-gui连接也没问题，operwrt连不上shadowsocks python版是什么原因呢，需要什么特殊设置嘛？

wickila, 2015-08-24:
首先确定是不是真的shadowsocks连不上服务器，如果telnet服务器对应的端口能通的话，那就是其他设置不对。你在openwrt里面安装的shadowsocks有3个部分——ss-local，ss-tunnel，ss-redir，不同的翻墙方案使用的ss服务也不一样，如果是至openwrt上启用的ss-turnnel来翻墙的话，还需要设置firewall的转发才能达到翻墙目的。

travelbob2005, 2015-08-24:
telnet端口可以的，你后面说的ss-local sstunnel ss-redir 我的路由器里没有看到，只有设置ss参数的，服务器，端口，密码，加密方式，google了半天，提到了operwrt的ss客户端是需要服务器段支持udp转发的，我安装的服务器端是shadowsocks-manyuser，Python版的，是不是需要手动开启udp转发？在哪里开启？

wickila, 2015-08-24:
你可以cat /etc/init.d/shadowsocks。就可以看到其实openwrt shadowsocks有三种服务模式的。我服务端用的就是python版本，没经过特殊设置。你可以加我Q：三四五六五八零零四，我看看能不能帮到你。

softwaredownload, 2015-12-23:
查看ss-server是否已经启动并且带有 -u启动参数
	ps ax | grep ss-server
	
如果启动正常，返回结果类似如下：
	
	/usr/bin/ss-server -c /etc/shadowsocks-libev/config.json -a root -u -f /var/run/shadowsocks-libev/shadowsocks-libev.pid
	
注意其中有-u。如果shadowsocks客户端启用了udp relay, 而服务端启动时不带-u参数，翻墙自然就失败了。

如果服务端启动里带　-u 翻墙还有问题，请检查其他设置

closed 2015-12-23 by softwaredownload