softwaredownload/openwrt-fanqiang #12: 这样添加国内ip这样添加可以解决有些某些国内站点走ss而不能正常访问的问题。

idonknown, 2015-11-29:
wget -O- 'http://ftp.apnic.net/apnic/stats/apnic/delegated-a
pnic-latest' | awk -F\| '/CN\|ipv4/ { printf("%s/%d\n", $4,
32-log($5)/log(2)) }' > cn_ip
这句命令获得一个apnic上的中国ip列表。
再用下面的命令添加进iptable规则，用的时间有10分钟左右。
# Ignore Asia IP address
for white_ip in `cat cn_ip`;
do
iptables -t nat -A SHADOWSOCKS_WHITELIST -d "${white_ip}" -j
 MARK --set-mark 1
done

softwaredownload, 2015-12-20:
Could submit a pull request?

softwaredownload, 2015-12-21:
Run Command on Ubuntu:

> wget -O- 'http://ftp.apnic.net/apnic/stats/apnic/delegated-apnic-latest' | awk -F | '/CN|ipv4/ { printf("%s/%d\n", $4,32-log($5)/log(2)) }' > cn_ip

Error:
Cannot write to ‘-’ (Broken pipe).

softwaredownload, 2015-12-23:
Added code blow in /usr/bin/shadowsocks-firewall:

    # Ignore China IP address
    for white_ip in `cat /etc/chinadns_chnroute.txt`;
    do
        iptables -t nat -A SHADOWSOCKS_WHITELIST -d "${white_ip}" -j MARK --set-mark 1
    done

closed 2015-12-23 by softwaredownload