softwaredownload/openwrt-fanqiang #14: DIR505 sysupgrade 后无法通过 wifi 连接，提示密码错误

leeyiw, 2015-12-23:
DIR505 sysupgrade 后无法通过 WIFI 连接，提示密码错误：
WIFI 名为 eastking-dir505，输入密码 icanfly9876 提示密码错误，请问正确的密码是多少？

small2huge, 2015-12-23:
我也是同样的问题，wifi密码错误。有线连接，浏览器输入192.168.1.1不能登录路由器。我刷openwrt.org下载的固件，是可以这样登录的。请给予指教，谢谢！

softwaredownload, 2015-12-23:
wsjdw,8181

softwaredownload, 2015-12-24:
wifi password for old binary: wsjdw,8181

small2huge, 2015-12-24:
谢谢，提供的密码登录成功。
刚才iphone手机ssh，修改为AP,
密码又变成icanfly9876
代码中如此。
可以上互联网了，但是不能翻墙。
怎么进一步设置呢？
谢谢指教。

softwaredownload, 2015-12-24:
按照教程仔细检查服务端和客户端相关设置，注意服务端启动参数中要有 -u 参数

leeyiw, 2015-12-24:
@softwaredownload 谢谢，请问新的固件在哪里下载？有入口吗？我通过 GitBooks 只能下载到旧版的固件。

softwaredownload, 2015-12-24:
新版正在编译，测试好会上传的

softwaredownload, 2015-12-24:
已经更新

small2huge, 2015-12-24:
我刷新了新版的固件，wifi连接dlink-55F4，输入密码icanfly9876，提示密码错误。旧的密码也不行。

softwaredownload, 2015-12-24:
用网线连接路由器和电脑，进入恢复模式登陆路由器，查看一下 /usr/bin/router 中的wifi key是不是 icanfly9876
我测试是可以正常登陆的

small2huge, 2015-12-24:
我对ubuntu不是很熟悉，我用的macbook pro，刚又刷新了固件，wifi名称和密码都正确了。可能是我刷错了固件。现在如何设置AP模式呢？也许ap模式才是最常用的模式，接上网线就可以翻墙了。您新编译的固件是router模式，如果能提供AP模式的固件就更好了，我不懂ubuntu命令行操作。还需要剪短开关吗？我的是dir501 a1欧版，我卸掉螺丝，拆开看了看，里面是一体成型的，拆不开。

leeyiw, 2015-12-24:
谢谢，解决了我的问题。

closed 2015-12-24 by leeyiw