softwaredownload/openwrt-fanqiang #16: DNS污染的问题还是没有彻底解决

kenchowcn, 2015-12-28:
大家好，我在我的OpenWRT路由器上安装了这个工具，我使用了几天，发现DNS污染的问题还是没有彻底解决；

我之前是用ChinaDNS，那个占用CPU资源比较高，而且有时候还是无法访问youtobe，facebook等。

然后，我发现这个[教程-配置OpenWrt路由器智能自动翻墙](https://github.com/softwaredownload/openwrt-fanqiang/blob/master/ebook/03.5.md)，通过ss-tunnel来转发DNS的请求到Google的DNS上，但是，这个Google的DNS不稳定，比如：ping 8.8.8.8，经常返回超时。

我觉得这种转发DNS请求的问题是一个一劳永逸的办法，我的想法是：

1. 在VPS上搭建一个dnsmasq的DNS服务器，绑定一个非标准的端口来监听DNS服务，比如：5050；
2. 通过ss-tunnel转发请求到VPS上；根据这个教程，我们只需要搭建dnsmasq的服务器就好了；

**我遇到的问题是，找了好多教程，无法把这个dnsmasq搭建好，有比较擅长的人可以写一个教程吗？**

BTW：我听说Docker就是用来解决配置繁琐的问题(好吧，是我觉得繁琐了)，我们是不是可以做一个把这些配置好的Docker的Container，我们的VPS直接安装一个Docker，然后，pull下来，配置一下IP和Port就好了，也不用分各种apt，yum管理器，还有安装依赖的问题了。

谢谢!
ken

softwaredownload, 2015-12-28:
dns查询是在shadowsocks服务端进行的，不可能受污染，不信你登陆shadowsocks服务端所在服务器ping 8.8.8.8试试

http://openwrtfanqiang.software-download.name/ebook/03.7.html

kenchowcn, 2015-12-28:
@softwaredownload 我查看了一下，我的ss-tunnel都是带了-u参数的，我在我openWRT下面接的windows设备ping 8.8.8.8，一般4次有3次超时。然后，我就觉得是DNS的问题，也许是别的问题，我继续检查看看。

谢谢！

closed 2015-12-28 by kenchowcn

jellyshen, 2015-12-28:
有提供防污染的DNS服务器