softwaredownload/openwrt-fanqiang #19: 请问修改SS端口账号密码的问题

PJovy, 2016-01-09:
VPS上装的python版本的SS，支持udp转发，我自己的SS开的端口不是1098，而是443，密码是其他，加密方式是rc4-md5，请问是不是直接修改了shadowsocks.json 上的服务器地址，服务器端口，SS密码，加密方式，然后修改防火墙部分为VPS的ip地址，然后重启就行了。以上就是我的所有设置

PJovy, 2016-01-09:
就是说我想修改SS的端口，加密方式，请问还需要改其他哪些文件，除了shadowsocks-firewall，shadowosocks.json应该还需要修改一下

softwaredownload, 2016-01-09:
这样就可以 了

softwaredownload, 2016-01-09:
当然拨号上网用户名密码要改，这个比较简单，网页界面就可以 设置

PJovy, 2016-01-09:
宽带上网这些都修改了，就是我按照那样把shadowsocks.json修改为我自己的设置了，然后把shadowsocks-firewall修改成VPS的ip地址，可是行不通诶。python版本的SS好像是支持udp转发吧

PJovy, 2016-01-09:
我把密码，加密方式改了就不行

PJovy, 2016-01-09:
只能按照端口1098，密码killgfw，加密方式aes-256-cfb  这样就可以成功，应该是还需要修改下其他的配置文件吧？

softwaredownload, 2016-01-09:
不需要修改其他文件了

ripples-alive, 2016-01-09:
@PJovy  你路由上 shadowsocks.json 的配置和服务器上只要保持一致应该都可以才对啊，咋会改个算法都不行，你是两边没一起改还是咋滴，或者你可以试着控制变量一项一项的改着看看？

softwaredownload, 2016-01-10:
是SS server port问题

PJovy, 2016-01-10:
服务器端口1098不能变吧

softwaredownload, 2016-01-10:
SS server pot可以变，不过不能用443

PJovy, 2016-01-10:
明白了，我之前用了443端口。还有，请问支持哪些加密方式，如果想用chacha20要怎么办呢

softwaredownload, 2016-01-10:
bf-cfb就很好用了，没必要用chacha20

ripples-alive, 2016-01-10:
@softwaredownload 啥时候有不能用 443 的说法的，刚试了下，我这里是可以的，我服务器端是 `shadowsocks-libev 2.4.0`

softwaredownload, 2016-01-10:
@ripples-alive 他的服务器上443端口已经被其他服务占用了。443是https的端口

PJovy, 2016-01-10:
我服务器的443端口正常使用啊，不给路由，通过电脑客户端使用正常de

softwaredownload, 2016-01-11:
那么可能你原先哪一步操作出现失误了，就像 @ripples-alive 说的 "或者你可以试着控制变量一项一项的改着看看"

softwaredownload, 2016-01-18:
openwrt shadowsocks 重启时，ss-redir或 ss-tunnel可能没有重启。验证方法如下：  
    /etc/init.d/shadowsocks stop
    ps  
运行上面命令后，如果还有ss-redir或ss-tunnel进程，说明stop没有完全成功，这时就要强制杀掉进程，或者重启路由器使修改的配置文件生效。
强制杀掉进程：  
    kill -9 pid

commit e5c4df1f (2016-01-19): fix #19 运行  有时并没有结束ss-redir 或ss-tunnel进程，这会导致修改  后需要重启路由器才能生效。加上  强制杀掉进程避免重启
closed 2016-01-19 by softwaredownload