softwaredownload/openwrt-fanqiang #42: 请教一个问题ss-tunnel的

willwhui, 2017-05-27:
我的路由器总是一阵阵的不能翻墙，找不到原因所在。
并且路由器不能翻墙的时候，通过手机在同一个局域网里面，用ss的安卓客户端是可以翻墙的。

我在您的文章：https://softwaredownload.gitbooks.io/openwrt-fanqiang/content/ebook/03.5.html
中写道：
    service_start /usr/bin/ss-tunnel -b 0.0.0.0 -c /etc/shadowsocks.json -l 3210 -L 8.8.8.8:53 -u 
我看了以下ss-tunnel的参数解释（https://github.com/shadowsocks/shadowsocks-libev/blob/master/doc/ss-tunnel.asciidoc）
感到困惑的一点是：
-c参数和-L参数一起用，是什么意图呢？
因为我这样理解的：
看起来-c对应的参数已经制订了远程服务器做域名解析
那么-L在指定路由器去8.8.8.8解析域名是什么意思呢？会不会从本地访问8.8.8.8去解析的时候，被墙污了呢？

willwhui, 2017-05-27:
您的文章中写道：“后面我们会配置shdowsocks-libev的本地客户端ss-tunnel转发本地端口3210的查询到远程自建服务器。”
如果是去8.8.8.8解析域名，那么就并没有到远程自建服务器查询呢？

willwhui, 2017-05-27:
您好，我仔细找了一些文章，目前认为-L参数是为ss服务器准备的，所以没有问题。
可能的问题是-u参数导致dns解析被劫杀。
最终，在你的方案基础上按照https://hong.im/2014/03/16/configure-an-openwrt-based-router-to-use-shadowsocks-and-redirect-foreign-traffic/
中提到的利用pdnsd将dns解析请求转为tcp，然后再交给ss-tunnel（去掉-u参数）
目前为止一切安好。

willwhui, 2017-05-28:
做了上述修改之后，有些网站还是会间歇性访问不了
进一步修改是将ss-tunnel使用的ss服务器改为和ss-redir不一样，即dns解析和内容数据传递采用不同的ss服务器。
目前工作挺稳定的。

softwaredownload, 2017-06-07:
不稳定的原因很多，教程中也提到过，服务器IP可能被GFW限制，此种情况换IP就能解决

你可以试一下重新启用udp，并且ss-tunnel和ss-redir都指向新的IP

willwhui, 2017-06-09:
你好，经过这些天的尝试，发现我之前提到的办法是有效的：
在路由器上增加一个pdnsd模块，将dns的udp请求包转换为tcp模式，通过ss-tunnel发到ss服务器，再由ss服务器转给8.8.8.8，是比较稳定的。
网上有人在提过某些运营商会不问缘由的直接丢弃udp包。
所以做这样的转换是比较有必要的。
我建议，您如果觉得可以的话，可以把pdnsd模块的装载也加入到教程中。
非常感谢您的教程，写的很好。
我这些天折腾的经验记录在我的gist里面[在Openwrt路由器上配置Shadowsocks](https://gist.github.com/willwhui/fe88628b263171a1a6c400ceda1c4794)

softwaredownload, 2017-06-09:
谢谢提供十分详细的记录

在我的许多测试中，遇到问题的多是：
1. IP被限制，此时换IP即解决
2. 速度慢或只能上外网，或只能上内网，原因可能是路由器的问题或连接的稳定性，重启路由器即解决

你说的udp问题还未遇到过

如果方便的话，可以写个PR

willwhui, 2017-06-11:
什么是PR？

softwaredownload, 2017-06-11:
PR，Pull Request
https://github.com/softwaredownload/openwrt-fanqiang/blob/master/ebook/09.4.md

Fork 本项目 ( https://github.com/softwaredownload/openwrt-fanqiang/fork )
创建你的分支 (git checkout -b my-new-feature)
提交你的改进 (git commit -am 'Add some feature')
Push到你的分支 (git push origin my-new-feature)
到github.com 创建 Pull Request

willwhui, 2017-06-13:
oh! 没反应过来，我是git新手，等我挤出时间就来做。我会尽力的。

softwaredownload, 2017-07-12:
DNS走udp的速度快，通过pdnsd走tcp的速度慢。如果有其他方法，建议还是不要走tcp，因为每次dns请求都慢了，打开每个需要翻墙的网页的速度都慢了

可以测试一下 更改udp查询的端口，如果有用，就不必要用pdnsd

我自己的测试中，翻墙不稳定的情况很少见，即使偶然半途出现状况，重启路由器一般能解决

closed 2017-07-23 by softwaredownload