softwaredownload/openwrt-fanqiang #45: 建议在增加转发udp流量的功能

tefiszx, 2017-06-21:
之所以建议增加udp流量转发的功能是因为如下原因：
1、youtube目前使用quic，udp协议是首选，目前路由器的ss配合iptables只能转发tcp流量，最新版的youtube的app会出现断流，而网页端播放目前没问题（chrome必须关闭quic）
2、很多游戏的网络版都使用udp协议。目前的转发存在一定的局限性
3、在转发tcp流量的基础上增加转发upd后，ss就成为准vpn了。应用面更广。

tefiszx, 2017-06-21:
已经找到youtube安卓客户端3秒断流的解决办法。方法如下：1、修改/etc/init.d/shadowsocks文件，把ss-redir 加-u参数启动；2、修改/usr/bin/shadowsocks-firewall文件，在防火墙规则中增加一条iptables规则，将443端口的upd重定向到shadowsocks监听的端口即可

tefiszx, 2017-06-21:
请本项目的创建者把该方法合并到项目中去。

softwaredownload, 2017-06-21:
感谢提供详细的信息

我在iPhone测试了 YouTube客户端，目前没有发现断流。我这里不方便测试你提出的改进

如果方便的话，能不能发个 Pull Request，我再把你的修改合并到教程中

具体步骤如下：

Fork 本项目 ( https://github.com/softwaredownload/openwrt-fanqiang/fork )
创建你的分支 (git checkout -b my-new-feature)
提交你的改进 (git commit -am 'Add some feature')
Push到你的分支 (git push origin my-new-feature)
到github.com 创建 Pull Request

softwaredownload, 2017-06-28:
已经更新，具体修改请见下面的提交，如有问题请反馈：

https://github.com/softwaredownload/openwrt-fanqiang/commit/28ce4f638c01447d40490c2c084d2e2dc2a5fc93

softwaredownload, 2018-10-12:
完全的 udp 转发还未实现，todo之一

closed 2020-04-04 by softwaredownload