softwaredownload/openwrt-fanqiang #53: D-Link DIR-505 (Firmware 2018-10-22) settings won't save

mapx, 2018-11-27:
It might be an issue of OpenWrt and turns out the device won't keep settings. Everything is reset after reboot regardless "Save & Apply" and "Save".

`openwrt-18.06.1-ar71xx-generic-dir-505-a1-squashfs-factory.bin` and `openwrt-18.06.1-ar71xx-generic-dir-505-a1-squashfs-sysupgrade.bin` are both tested.

softwaredownload, 2018-11-28:
/etc/rc.local will be executed once the system init finished, it might be an issue of OpenWrt if the settings you changed does not relate to this file, in this case, you need to change settings via command line

softwaredownload, 2018-11-28:
/etc/rc.local will be executed once the system init finished, it might be an issue of OpenWrt if the settings you changed are not related to this file, in this case, you need to change the settings from command line